openlumi/openlumi.github.io #4: ZiGate don't start on port 9440

emwu88, 2020-12-23:
After clean install system (mfgtools-19.07.5-20201217), and install domoticz (https://github.com/openlumi/xiaomi-gateway-openwrt/blob/master/domoticz.md) plugin don't start on port 9440.

root@OpenWrt:~ netstat -tulpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1497/uhttpd
tcp        0      0 192.168.8.102:53        0.0.0.0:*               LISTEN      1312/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      1312/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1345/dropbear
tcp        0      0 127.0.0.1:6010          0.0.0.0:*               LISTEN      1976/dropbear
tcp        0      0 :::6144                 :::*                    LISTEN      1834/domoticz
tcp        0      0 :::6600                 :::*                    LISTEN      1700/mpd
tcp        0      0 :::8080                 :::*                    LISTEN      1834/domoticz
tcp        0      0 :::80                   :::*                    LISTEN      1497/uhttpd
tcp        0      0 fe80::6690:c1ff:fe77:9a59:53 :::*                    LISTEN      1312/dnsmasq
tcp        0      0 ::1:53                  :::*                    LISTEN      1312/dnsmasq
tcp        0      0 :::22                   :::*                    LISTEN      1345/dropbear
udp        0      0 192.168.8.102:53        0.0.0.0:*                           1312/dnsmasq
udp        0      0 127.0.0.1:53            0.0.0.0:*                           1312/dnsmasq
udp        0      0 fe80::6690:c1ff:fe77:9a59:53 :::*                                1312/dnsmasq
udp        0      0 ::1:53                  :::*                                1312/dnsmasq

2020-12-23 06:25:28.125 Active notification Subsystems: fcm, http (2/13)
2020-12-23 06:25:28.245 Starting shared server on: :::6144
2020-12-23 06:26:28.801 (zigate) DomoticzVersion: 2020.2
2020-12-23 06:31:43.951 (zigate) Debug Mode: 14, We do recommend to leave Verbors to None
2020-12-23 06:31:43.952 (zigate) 'HardwareID':'2'
2020-12-23 06:31:43.953 (zigate) 'HomeFolder':'/etc/domoticz/plugins/Domoticz-Zigate/'
2020-12-23 06:31:43.954 (zigate) 'StartupFolder':'/usr/share/domoticz/'
2020-12-23 06:31:43.955 (zigate) 'UserDataFolder':'/etc/domoticz/'
2020-12-23 06:31:43.956 (zigate) 'Database':'/etc/domoticz/domoticz.db'
2020-12-23 06:31:43.957 (zigate) 'Language':'en'
2020-12-23 06:31:43.958 (zigate) 'Version':'4.11'
2020-12-23 06:31:43.959 (zigate) 'Author':'zaraki673 pipiche38'
2020-12-23 06:31:43.959 (zigate) 'Name':'zigate'
2020-12-23 06:31:43.960 (zigate) 'Address':'0.0.0.0'
2020-12-23 06:31:43.961 (zigate) 'Port':'9999'
2020-12-23 06:31:43.962 (zigate) 'SerialPort':'/dev/ttymxc1'
2020-12-23 06:31:43.963 (zigate) 'Key':'Zigate'
2020-12-23 06:31:43.964 (zigate) 'Mode1':'DIN'
2020-12-23 06:31:43.965 (zigate) 'Mode3':'False'
2020-12-23 06:31:43.966 (zigate) 'Mode4':'9440'
2020-12-23 06:31:43.967 (zigate) 'Mode6':'14'
2020-12-23 06:31:43.968 (zigate) 'DomoticzVersion':'2020.2'
2020-12-23 06:31:43.969 (zigate) 'DomoticzHash':'0'
2020-12-23 06:31:43.970 (zigate) 'DomoticzBuildTime':'1970-01-01 00:00:00'
2020-12-23 06:31:43.970 (zigate) Device count: 0
2020-12-23 06:31:43.971 (zigate) DomoticzVersion: 2020.2

devbis, 2021-01-08:
It could take several minutes to make the plugin's web interface to start. 
You could double-check the permissions for `/etc/domoticz` folder or check the logs for the plugin in debug mode.

pipiche38, 2021-03-03:
the other reason why you get this error, is due to no communication established between the plugin and the ZiGate host .
So cross check that the IP link between the domoticz host and the Zigate host is up

closed 2021-04-18 by devbis