openlumi/openlumi.github.io #8: rc.local now empty

Sunnyboy2284, 2021-02-07:
Hi i had a conflict during programming, now my rc.local is empty and i have only access via UART, can  anybody send me the rc.local content. thx.

devbis, 2021-02-07:
If you are using OpenWrt you can delete /overlay/upper/etc/rc.local

```
rm -f  /overlay/upper/etc/rc.local
reboot
```

That will revert the file to the initial content.

Sunnyboy2284, 2021-02-07:
hi thx, 

no actually i dont have open wrt, i did now copy rc.local.mi to rc.local now it s working but i did it two times before it works .
By the way , if i use open wrt , can i use mi home and gateway with this firmware or can I only use it in Home Assistant or as standalone?

thanks for  fast response.

devbis, 2021-02-07:
This instruction is about installing OpenWrt. Stock OS doesn't support customization.

You can use the gateway as a standalone device. Install zigbee2mqtt + domoticz + z2m plugin for domoticz to manage devices or zesp32. This software is mentioned in the manual.

closed 2021-02-07 by devbis