openlumi/openlumi.github.io #12: Disabling OpenWrt default access point from LUCI shutting down all wifi access.

dssoftgh, 2021-02-26:
After installation we need to setup wireless access, for example, connect to the main router as client, and disable "maintenance" OpenWrt after it. But clicking on the "disable" button actually disabling both radios (client and ap) and preventing from login.
Here /config/wireless:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'soc0/soc/2100000.aips-bus/2190000.usdhc/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'wwan'
        option mode 'sta'
        option ssid 'MYAP'
        option encryption 'psk2'
        option key 'MYPASSWORD'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'soc0/soc/2100000.aips-bus/2190000.usdhc/mmc_host/mmc0/mmc0:0001/mmc0:0001:1+1
        option disabled '1' <---- problem is here

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option disabled '1'

```
LUCI insert "option disabled" to both sections, default_radio1 and radio1 itself, and after applying both radio0 and radio1 shutted down because of same path.
Solution: remove first "option disabled" from radio1 config.

Rus: В общем, если попытаться выключить дефолтную точку доступа OpenWrt то вайфай на устройстве выключается полностью, даже если настроен доступ клиентом.

devbis, 2021-02-26:
You can avoid this if you follow the steps to set up WiFi in the instruction.
If problems occur, you can reboot the gateway and start from scratch. Or do reset by pressing 20 seconds on the button on the gateway.

There is no other solution while the wifi drivers from Realtek are buggy and can cause issues.

closed 2021-02-26 by devbis

dssoftgh, 2021-02-26:
OpenWrt "Factory reset" feature (pressing button) does not working too, but this is another issue. I spend half of hour today trying to reset, but ended with serial console.