openlumi/openlumi.github.io #13: WireGuard

dvv, 2021-02-27:
```

devbis, 2021-02-27:
You have to compile OpenWrt by yourself and enable the required kernel modules in the firmware.
Otherwise, some kernel modules may cause issues.

Sources for the OpenWrt 19.07 are located here: https://github.com/openlumi/openwrt/tree/lumi-19.07

closed 2021-02-27 by devbis

dvv, 2021-02-27:
On my router:
```
$ uname -a
Linux home 4.14.167 #0 SMP Wed Jan 29 16:05:35 2020 mips GNU/Linux
```
gives
```
$ lsmod | egrep '(wire|tunnel)'
ip6_udp_tunnel          1600  1 wireguard
udp_tunnel              2432  1 wireguard
wireguard              90560  0
```
so all the required kernel modules already _do exist_ in this firmware.

It's a matter of some missing kernel feature. I believe `select DST_CACHE` should go ( mentioned [https://lore.kernel.org/patchwork/patch/880428/](https://lore.kernel.org/patchwork/patch/880428/) )

I can not afford building the world. May some be interested.

devbis, 2021-02-27:
The modules that exist in firmware depends on the config for building.
It may include some modules, may not.
We tried to keep the image small and did not include most of the kernel modules that are not required for basic gateway usage.
Here is the config I used for the build: https://github.com/openlumi/openwrt/blob/lumi-19.07/config_xiaomi_lumi

dvv, 2021-02-27:
I see.

Look at what we've got:

```
$ modinfo wireguard
module:         /lib/modules/4.14.221/wireguard.ko
intree:         Y
alias:          net-pf-16-proto-16-family-wireguard
alias:          rtnl-link-wireguard
license:        GPL v2
depends:        udp_tunnel,ip6_udp_tunnel

$ modinfo udp_tunnel
module:         /lib/modules/4.14.221/udp_tunnel.ko
license:        GPL
depends:

$ modinfo ip6_udp_tunnel
module:         /lib/modules/4.14.221/ip6_udp_tunnel.ko
license:        GPL
depends:
```

So I conclude that the missing symbols are to be in the shipped kernel per se.

dvv, 2021-02-27:
@devbis  Am I right that the following will give me exact config that you have?

```sh
git clone --depth 1 -b lumi-19.07 https://github.com/openlumi/openwrt
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a
cp config_xiaomi_lumi .config
make menuconfig
```

dvv, 2021-02-28:
Built, `sysupgrade`-d, got a brick... )
Suspecting WiFi drivers

devbis, 2021-02-28:
Oh, you've missed one essential step.

files_xiaomi_lumi should be renamed to files or make a symlink with files name.

You could try to restore availability by copying the content to /lib/* files through serial console.

dvv, 2021-03-05:
Thank you. Revived. That comment ^^^ should go to readme imho.