openlumi/openlumi.github.io #15: Serial connection seems to broke after a few seconds

danielk117, 2021-04-01:
Hey guys,

I just bought a ZHWG11LM and want to flash it with your OpenWRT.
I tried to gain root access, following your guide. So I added my CH340 to GND, TX and RX.

When starting the device, I can see it booting. I'm able to interupt it by pressing any key and after this i see the `=>`. 
I'm able to type or paste the `setenv ...` for around 1 second and than the device doesn't react on my input anymore. Also, I can no longer read the output. There are just some unknown symbols (like if you use a wrong baudrate, but I use baudrate 115200).

If I'm fast, I'm able to paste the whole `setenv` command and press enter. The last thing I could read is `NAND read ...`. If I don't interrupt the boot, the same thing happen. Around 1 second after the short countdown there are just unreadable symbols. 
I tried it with `screen` and `minicom` an macOS and PuTTY on Windows. All with the same CH340 adapter. Same thing.
The CH340 should be ok, just flashed a nodeMCU with it a week ago.

Any ideas what I'm doing wrong? 😞 
If you want to see a screenshot of my output, I paste it later that day when back at home.

danielk117, 2021-04-01:
```
U-Boot 2016.03 (Nov 07 2017 - 20:11:55 +0800)

CPU:   Freescale i.MX6ULL rev1.0 528 MHz (running at 396 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 41C
Reset cause: POR
Board: Lumi international Gateway
I2C:   ready
DRAM:  256 MiB
NAND:  256 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

Display: TFT43AB (480x272)
Video: LCDIF@0x21c8000 is fused, disable it
In:    serial
Out:   serial
Err:   serial
Net:   Ethernet@0x20b4000 is fused, disable it
Board Net Initialization Failed
No ethernet found.

## Enter main_loop() Now##
Normal Boot
Hit any key to stop autoboot:  0
=>
```

devbis, 2021-04-03:
Check your soldering, wires, and uart adapter. One of these seems inapropriate

danielk117, 2021-04-03:
Tried new wires and soldered them. Same problem.
Ordered a new adapter now...

danielk117, 2021-04-06:
New adapter, same result.
Since it was never my intention, I did not try to connect with Mi Home app or HomeKit before. Tried it now and it doesn‘t work. 
Resetting doesn’t work either. 

So soldering, wires and adapter seems to be ok. The gateway is garbage... ☹️

devbis, 2021-04-06:
There can be many reasons that lead to this behavior. Try another desktop/laptop, try to shorten the wires.

danielk117, 2021-04-06:
Read my first post. I was trying on windows and macOS before. Wires are already short. 

I‘m not able to pair it with the xiaomi app. Resetting with holding the button for 10s doesn‘t work. 
I could try any wire-adapter-laptop-combonation, the device itself seems to be damaged. 
I will replace it by a new one...

devbis, 2021-04-06:
Okay, that is a good decision. There is a large percentage of broken devices.

closed 2021-04-06 by devbis