openlumi/openlumi.github.io #16: Device bricked "itself". No option to take action via uart.

broda1125, 2021-04-02:
Hi there,

My aqara hub turned into openlumi mqtt router stopped working itself - came back from work and it was dead.
I can connect via uart and run basic commands but I have no idea how to bring it back to life... after booting can't enter any command, no permissions to run recovery_mode (failsafe). Below dump from console, could you help somehow? Cheers!

`U-Boot 2016.03 (Nov 07 2017 - 20:11:55 +0800)

CPU:   Freescale i.MX6ULL rev1.1 528 MHz (running at 396 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 30C
Reset cause: POR
Board: Lumi international Gateway
I2C:   ready
DRAM:  256 MiB
NAND:  256 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

Display: TFT43AB (480x272)
Video: LCDIF@0x21c8000 is fused, disable it
In:    serial
Out:   serial
Err:   serial
Net:   Ethernet@0x20b4000 is fused, disable it
Board Net Initialization Failed
No ethernet found.

## Enter main_loop() Now##
Normal Boot
Hit any key to stop autoboot:  0
=> boot

NAND read: device 0 offset 0x300000, size 0x700000
 7340032 bytes read: OK

NAND read: device 0 offset 0xa00000, size 0x100000
 1048576 bytes read: OK
Kernel image @ 0x80800000 [ 0x000000 - 0x2abd58 ]
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300973c
Modify /soc/aips-bus@02000000/tsc@02040000:status disabled
Modify /soc/aips-bus@02100000/lcdif@021c8000:status disabled
Modify /soc/aips-bus@02000000/ethernet@020b4000:status disabled
Modify /soc/aips-bus@02100000/usb@02184200:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/sai@0202c000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/sai@02030000:status disabled
Modify /soc/aips-bus@02100000/serial@021f4000:status disabled
Modify /soc/aips-bus@02100000/serial@021fc000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/serial@02018000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/serial@02024000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f0000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f4000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f8000:status disabled
Modify /soc/aips-bus@02000000/pwm@020fc000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/ecspi@02010000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/ecspi@02014000:status disabled
Modify /soc/aips-bus@02100000/i2c@021a8000:status disabled
Modify /soc/aips-bus@02100000/i2c@021f8000:status disabled
Modify /soc/aips-bus@02000000/gpt@020e8000:status disabled
Modify /soc/aips-bus@02000000/epit@020d4000:status disabled
ft_system_setup for mx6
No PMIC found!

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.215 (bis@betelgeuze) (gcc version 7.5.0 (OpenW                                                                                                                                                                             rt GCC 7.5.0 r11232+24-2a3dbded93)) #0 SMP Tue Jan 19 13:10:02 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio                                                                                                                                                                             n cache
[... 153 lines cut ...]
[    5.607713] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    5.647944] Button Hotplug driver version 0.4.1
[    5.654818] Error: Driver 'gpio-keys' is already registered, aborting...
[    5.669826] usbcore: registered new interface driver usb-storage
[    5.679727] Error: Driver 'gpio-keys' is already registered, aborting...
[    5.687622] kmodloader: 1 module could not be probed
[    5.693123] kmodloader: - gpio-button-hotplug - 0
[    5.708487] init: - preinit -
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[   10.268063] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 797
[   10.348520] UBIFS (ubi0:1): recovery needed
[   10.533798] UBIFS (ubi0:1): recovery completed
[   10.538411] UBIFS (ubi0:1): UBIFS: mounted UBI device 0, volume 1, name "root                                                                                                                                                                             fs_data"
[   10.546331] UBIFS (ubi0:1): LEB size: 126976 bytes (124 KiB), min./max. I/O u                                                                                                                                                                             nit sizes: 2048 bytes/2048 bytes
[   10.556303] UBIFS (ubi0:1): FS size: 230969344 bytes (220 MiB, 1819 LEBs), jo                                                                                                                                                                             urnal size 11554816 bytes (11 MiB, 91 LEBs)
[   10.567221] UBIFS (ubi0:1): reserved for root: 4952683 bytes (4836 KiB)
[   10.573885] UBIFS (ubi0:1): media format: w4/r0 (latest is w5/r0), UUID B480E                                                                                                                                                                             49D-B0FF-4AFF-8AB2-D7EA24210D5F, small LPT model
[   10.601297] mount_root: switching to ubifs overlay
/etc/preinit: line 358: cat: Permission denied
[   10.677452] urandom-seed: Seeding with /etc/urandom.seed
/etc/preinit: line 12: cat: Permission denied
/etc/preinit: line 20: uci: Permission denied
/etc/preinit: line 120: grep: Permission denied
`

devbis, 2021-04-03:
You can try the following to reset the overlay:

### 1.Enter failsafe mode 

After prompt appears :

Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level

Press [f] and then press enter.

### 2. Print the following lines in the failsafe prompt:
```sh
ubirmvol /dev/ubi0 -Nrootfs_data
ubimkvol /dev/ubi0 -Nrootfs_data -m
```

### 3. Reboot and configure the gateway

broda1125, 2021-04-03:
Thanks a lot! It worked!
Is it possible to use gateway as router and yet be able to control its lights, button click etc?
And what exactly happened here with my gw that I couldn't run it?

devbis, 2021-04-06:
> Is it possible to use gateway as router and yet be able to control its lights, button click etc?

Yes, you can write router firmware in the zigbee chip and use other software on the gateway for your own purposes. 

> And what exactly happened here with my gw that I couldn't run it?

I have no idea, some files were corrupted.

closed 2021-04-18 by devbis