openlumi/openlumi.github.io #17: failed to update firmware

nightmare04, 2021-04-18:
try to install OpenWrt...

CONNECTED(00000003)
write:errno=104
---
no peer certificate available
---
No client certificate CA names sent
---
SSL handshake has read 0 bytes and written 307 bytes
---
New, (NONE), Cipher is (NONE)
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : 0000
    Session-ID:
    Session-ID-ctx:
    Master-Key:
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 2829
    Timeout   : 300 (sec)
    Verify return code: 0 (ok)
---

devbis, 2021-04-18:
These lines don't make sense without context. What did you do? From what exactly did you update?
Did you try flashing over USB?

devbis, 2021-04-22:
Fixed in 63cacaf7001bf0a616c1e58bcec8332cb8fe3c94

closed 2021-04-22 by devbis