openlumi/openlumi.github.io #18: [REQUEST] Zeroconf or SSDP automatic network discovery compatibility with Home Assistant ZHA integration

Hedda, 2021-04-22:
Please consider adding automatic network discovery of this so can be discovered by Home Assistant’s ZHA (Zigbee) integration:

https://community.home-assistant.io/t/zha-automatic-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-devices-that-support-zeroconf-or-ssdp/293300

Support for Zeroconf network discovery of Tube's ESPHome based Zigbee Gateways was recently added to the ZHA integration: 

https://github.com/home-assistant/core/pull/48420

Note that Home Assistant already have integration support to add Zeroconf and/or SSDP automatic network discovery methods.

devbis, 2021-04-29:
Hey Hedda,
this is just an instruction to install OpenWrt and some software on the gateway. 
I don't have an idea what are you talking about.
If you can add support for zeroconf to OpenWrt on the gateway, your PRs are welcome in the neighbor OpenWrt repo.
The default usage that we recommend is zigbee2mqtt installed on the gateway

Hedda, 2021-04-30:
> I don't have an idea what are you talking about.

This would be for using the gateway hardware just as (dumb) Zigbee serial port adapter over a serial-to-ip proxy/bridge server service (like example ser2net) in order to provide remote access to external Zigbee applications running on another computer.

That is, not running Home Assistant on Zigbee2MQTT directly on the gateway but instead only using it as a remote adapter.

This is a relatively common method for getting a "network-attached Zigbee adapter" instead of local Zigbee adapter via USB as the benefit is that you can run Home Assistant on Zigbee2MQTT on another computer (like a virtual machine on your NAS and get all the benefits that entail) and still be able to place the gateway anywhere in your house as a remote Zigbee coordinator.

Popular projects/products that provide this type of Zigbee adapter serial access over TCP/IP include Tube’s Zigbee Gateways, ZiGate WiFi Pack, hacked Tuya TYGWZ-01 / Lidl Silvercrest Smart Gateway, and most popular is hacking ITead Sonoff ZBBridge:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

https://paulbanks.org/projects/lidl-zigbee/ha.html

https://github.com/tube0013/tube_gateways

https://zigate.fr/produit/zigate-pack-wifi-v1-3/ -> https://github.com/fairecasoimeme/ZiGate-WiFi

Connect to a remote Zigbee adapter over IP via ser2net for the same purpose is even described in Zigbee2MQTT documentation:

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

See related Zeroconf discussion for Home Assistant for more information https://community.home-assistant.io/t/zha-automatic-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-devices-that-support-zeroconf-or-ssdp/293300

> If you can add support for zeroconf to OpenWrt on the gateway, your PRs are welcome in the neighbor OpenWrt repo.

I would have posted the request to https://github.com/openlumi/openwrt instead of but it is enabled to receive issues.

devbis, 2021-04-30:
After a fresh install, you receive an almost empty OpenWrt system. 
There is no ser2net installed and there are other ways to use the zigbee chip besides wifi bridge to an external system.
Moreover, the recommended way is to install z2m directly on the gateway.

To achieve your proposal, there should be preinstalled and preconfigured ser2net along with software for zeroconf (Avahi server?)
I suppose thorough research should be done to add this functionality.
If you have written an instruction how to implement it with xiaomi gateway, contribute it as a PR please.

Hedda, 2021-09-24:
FYI, there is now a working proof-of-concept how this is now supported by Tube's Zigbee gateways (based on ESPHome firmware):

https://www.home-assistant.io/integrations/zha#discovery-via-usb-or-zeroconf

https://github.com/tube0013/tube_gateways

So today it is probably best to see actual example config for ESPHome as in Tube's Zigbee Gateway as that is the reference:

https://github.com/tube0013/tube_gateways/blob/main/V2_tube_zb_gw_cc2752p2/ESPHome/tube_zb_gw_cc2652p2v2.yml

```
zeroconf:
  - service: tubes_zb_gw
    protocol: tcp
    port: 6638
    txt:
      version: 1.0
      radio_type: znp
      baud_rate: 115200
      data_flow_control: software
```

Again, support for Tube's Zigbee Gateway was initially added to Home Aassistant core for ZHA support in https://github.com/home-assistant/core/pull/48420

As I believe to then whitelist Zeroconf for more gateways and/or radio types in HA's zeroconf as well as for ZHA need to do PR for:

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py

```
 "_esphomelib._tcp.local.": [
        {
            "domain": "zha",
            "name": "tube*"
        }
    ],
```

and

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json

```
"zeroconf": [
    {
      "type": "_esphomelib._tcp.local.",
      "name": "tube*"
    }
  ],
```
and

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/config_flow.py

```
async def async_step_zeroconf(self, discovery_info: DiscoveryInfoType):
        """Handle zeroconf discovery."""
        # Hostname is format: livingroom.local.
        local_name = discovery_info["hostname"][:-1]
        node_name = local_name[: -len(".local")]
        host = discovery_info[CONF_HOST]
        device_path = f"socket://{host}:6638"

        if current_entry := await self.async_set_unique_id(node_name):
            self._abort_if_unique_id_configured(
                updates={
                    CONF_DEVICE: {
                        **current_entry.data.get(CONF_DEVICE, {}),
                        CONF_DEVICE_PATH: device_path,
                    },
                }
            )

        # Check if already configured
        if self._async_current_entries():
            return self.async_abort(reason="single_instance_allowed")

        self.context["title_placeholders"] = {
            CONF_NAME: node_name,
        }

        self._device_path = device_path
        self._radio_type = (
            RadioType.ezsp.name if "efr32" in local_name else RadioType.znp.name
        )

```

Then DNS TXT records should be used to pass along recommended settings parameters config flow to the ZHA domian in Home Assistant:

```
version=1.0
radio_type=ezsp
baud_rate=value
data_flow_control=software
```

I believe that is advertised for Zeroconf config in this format or similar:
```
zha_ezsp_zeroconf  _ezsp._tcp  local
   hostname = [zha_ezsp_zeroconf.local]
   port = [8080]
   protocol = [tcp]
   service = tubes_zb_gw
   txt = ["version=1.0"]
   txt = ["radio_type=ezsp"]
   txt = ["baud_rate=value"]
   txt = ["data_flow_control=software"]
```
(use e.g. avahi-browse -r -a to see this)

As can see, you will need one DNS TXT Record for each attribute and value that is to be passed along to HA's ZHA integration.

Again, Zeroconf DNS TXT records can also be used to pass along info about hostname, versions, location, MAC address, etc..

Hedda, 2021-09-24:
> After a fresh install, you receive an almost empty OpenWrt system.
> There is no ser2net installed and there are other ways to use the zigbee chip besides wifi bridge to an external system.
> Moreover, [...]

By the way, it looks like an "Zigbee Home Automation Mode" is currently being implemented as an optional feature by AlexxIT in XiaomiGateway3 which otherwise is a project that has the same concept as openlumi but is made for ZNDMWG03LM and ZNDMWG02LM 

https://github.com/AlexxIT/XiaomiGateway3

https://github.com/AlexxIT/XiaomiGateway3#zigbee-home-automation-mode

devbis, 2021-09-24:
The installation doesn't provide ANY services by default. No z2m, no serial-over-ip, no other services except raw openwrt system. It cannot be used with ZHA without installing extra software like ser2net.
A user CAN install programs on it and if it provides zeroconf functionality, it could be used by home assistant.
Zeroconf is a tool to find services in the network, not devices. 

In comparison to XiaomiGateway3 the last one uses the orignal system with services.

closed 2022-08-02 by devbis