openlumi/openlumi.github.io #20: Not able to flash JN51XX chip

galizki, 2021-05-22:
Hi, thank you for you efforts to bring this project to live, it is really great!

However, after installing OpenLumi within full soldering via UART + USB and installing 'Version 19.07.7 2021-02-24' I am having the issue with updating firmware JN51XX Zigbee chip.

The reason why I am trying to update the firmware is that I have not done it before installing zigbee2mqtt and failed to make it receiving commands from Ikea Zigbee devices , however all xiaomi buttons and sensors worked just fine. I assume that it is because I have not updated the firmware of the JN51XX chip  and it is how the story begins..

If I launch firmware update within web interface, by filling in the URL and clicking OK, i got:

```
Flash /tmp/firmware.bin
JennicModuleProgrammer Version: 0.11 (r1234)
Logging Serial port opened at iInitialSpeed (= 38400) to /tmp/flasher.log
Setting baudrate for port 3 to 1000000
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (2)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (1)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (0)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log

```

I am getting the same result by entering the command manually in the console: 

```

root@XiaomiGateway:~# wget https://github.com/openlumi/ZiGate/releases/download/snapshot-20201201/ZigbeeNodeControlBridge_JN5169_FULL_FUNC_DEVICE_31e_115200.bin -O /tmp/ziga
te.bin 
Downloading 'https://github.com/openlumi/ZiGate/releases/download/snapshot-20201201/ZigbeeNodeControlBridge_JN5169_FULL_FUNC_DEVICE_31e_115200.bin'
Connecting to 140.82.121.3:443
Redirected to /302286515/44627800-415d-11eb-9070-e15e4e74c14e?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210522%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210522T170937Z&X-Amz-Expires=300&X-Amz-Signature=d8c619150901cd9485a61224df8c1288886c4fab54cb6bb7f2433e0f89462c19&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=302286515&response-content-disposition=attachment%3B%20filename%3DZigbeeNodeControlBridge_JN5169_FULL_FUNC_DEVICE_31e_115200.bin&response-content-type=application%2Foctet-stream on github-releases.githubusercontent.com
Writing to '/tmp/zigate.bin'
/tmp/zigate.bin      100% |*******************************|   221k  0:00:00 ETA
Download completed (226944 bytes)
root@XiaomiGateway:~# jnflash /tmp/zigate.bin
Flash /tmp/zigate.bin
JennicModuleProgrammer Version: 0.11 (r1234)
Logging Serial port opened at iInitialSpeed (= 38400) to /tmp/flasher.log
Setting baudrate for port 3 to 1000000
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (2)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (1)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log
Error setting (bootloader) baudrate to iProgramSpeed (1000000) (0)
Logging Error setting (bootloader) baudrate to iProgramSpeed (= 1000000) to /tmp/flasher.log

```

I have also played with another options to flash JS51XX and only the following gives another result (but it is also failed):

```
root@XiaomiGateway:~# jnprogrammer -v -s /dev/ttymxc1  -I 115200 -P 115200 -f ./zigate_JN5169_115200_31e.bin 
JennicModuleProgrammer Version: 0.11 (r1234)
Logging Serial port opened at iInitialSpeed (= 38400) to /tmp/flasher.log
Error reading module information - check cabling and power
Logging Error reading module information - check cabling and power to /tmp/flasher.log
```

I have also tried soft_reset and erase_pdm options of jntool and it does not help. 

I am not sure if this issue related exactly to openlumi, but I really do not understand where the problem is. Please please advice.

devbis, 2021-05-23:
It seems your zigbee port is busy. Stop zigbee2mqtt or any other software that may open and block the port and try again.

galizki, 2021-05-23:
Thank you for your help! 
Launching `/etc/init.d/zigbee2mqtt stop` helped to resolve the flashing issue. 

It is still issues with capturing commands from Ikea controls, but this question is out of this topic. Thanks a lot.

closed 2021-05-23 by galizki