openlumi/openlumi.github.io #23: Dropbear SSH can not start

devoll, 2021-07-02:
Hi! First of all, thank you for your project!

I have Aqara Gateway 2 (ZHWG11LM) and trying to get root by your instructions.
I have a problem with dropper ssh. It's not start and I see the next logs:

```
...
==========================================
======== ^_^ start miot hkbridge =========
==========================================
Starting Dropbear SSH server: fac_test bulid time:04:06:52 Oct 19 2018
gobal_cmd_list size 3204
Input cmd:Pseudo-terminal will not be allocated because stdin is not a terminal.
dropbearkey: Could not resolve hostname rsa: Name or service not known
```

As I understand, ssh service didn't run, because I cannot connect to it.

This is start gateway logs (with some versions of software and hardware) if it help you:
```
U-Boot 2016.03 (Nov 07 2017 - 20:11:55 +0800) 

CPU:   Freescale i.MX6ULL rev1.0 528 MHz (running at 396 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 41C
Reset cause: POR
Board: Lumi international Gateway
I2C:   ready
DRAM:  256 MiB
NAND:  256 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

Display: TFT43AB (480x272)
Video: LCDIF@0x21c8000 is fused, disable it
In:    serial
Out:   serial
Err:   serial
Net:   Ethernet@0x20b4000 is fused, disable it
Board Net Initialization Failed
No ethernet found.
```

devbis, 2021-07-02:
Some Aqara firmware binaries include dropbear compiled without ssh server. It can be used only as a client.
To fix this issue you can replace `dropbearmulti` with the one from xiaomi dgnwg05lm firmware. It is already uploaded to github.

I suppose you can use the following commands under root user on your gateway:

```
mv /usr/sbin/dropbearmulti /usr/sbin/dropbearmulti.backup
echo -e "GET /openlumi/openlumi.github.io/master/files/dropbearmulti HTTP/1.0\nHost: raw.githubusercontent.com\n" | openssl s_client -quiet -connect raw.githubusercontent.com:443 -servername raw.githubusercontent.com 2>/dev/null | sed '1,/^\r$/d' > /usr/sbin/dropbearmulti
chmod +x /usr/sbin/dropbearmulti
```

devoll, 2021-07-02:
Thank you so much! It's alive!

closed 2021-07-02 by devoll