openlumi/openlumi.github.io #30: Alarm mode support

giltabibian, 2021-11-15:
Hi! Thanks for this great guide! 
I was able to connect to uart and flash the firmware on my DGNWG05LM gateway. Still wondering witch firmware to install.
Cant find any mention of alarm mode support? Is it possible to use it? 
I understand that i cant use both xiaomi mi account and ZHA/z2mqtt. My purpose is to use both the zigbee sensors and the hub alarm and light. (using home assistant on raspberry pi 4)

devbis, 2021-11-15:
Alarm mode is not implemented. You can use https://github.com/openlumi/lumimqtt to flash light through MQTT and MPD HA integration to play sounds. Also, you can implement a custom shell command with MQTT events.

devbis, 2021-12-17:
I don't think that stock firmware is customizable in any way

giltabibian, 2021-12-17:
Ok. And does the stock fw serving as a repeater to all zegbee messages it receives?

devbis, 2021-12-17:
I don't know, doubt it :)

closed 2021-12-23 by devbis