openlumi/openlumi.github.io #31: flash backup image from uboot

giltabibian, 2021-11-15:
Hi,
is there any easy way to restore to back image? (lumi_stock.tar.gz)
using mfgtools is a little overhead because of soldering the usb pins.
it seems like uboot doesn't have an ethernet port, so using tftpboot / nfs to load an image and burn it is not an option right?
currently, i have openwrt, can't i use the web interface to flash the backup firmware?

devbis, 2021-11-16:
Nope, there is no way to restore backup without soldering usb.
In theory, it is possible, but prevoius attempts led to read-only fs and usb soldering was needed anyway.
You can investigate this way and post your findings.

giltabibian, 2021-11-16:
Thanks for you feedbacks on my open issues. Highly appreciated!

closed 2021-12-02 by devbis