openlumi/openlumi.github.io #32: HA Integration support with DGNWG05LM

giltabibian, 2021-11-21:
Can i use this integration in home assistant:
https://github.com/AlexxIT/XiaomiGateway3

Using DGNWG05LM gateway and followed those steps to gain root and wifi config. Now i can access remotely without uart:
https://openlumi.github.io/gain_root.html

devbis, 2021-11-21:
No, you cannot. It is for another model of gateway

giltabibian, 2021-11-21:
@devbis why not? What is the difference from other gateway?
The integration based on telnet / ssh with root permissions?
So after connecting to UART and fix those for DGNWG05LM model i guess it should work right?
Can you list what are the differences that make it impossible to use this model with this integration?

devbis, 2021-11-23:
It has completely different hardware (and firmware). Read the manual on the component's page.

closed 2021-11-24 by devbis

giltabibian, 2021-11-24:
Thanks for the informative comments 😊