openlumi/openlumi.github.io #33: SSH does not work

ralocar, 2021-11-22:
Hi,
I have won root with Serial port but ssh does not work.
I tried with:

mv /usr/sbin/dropbearmulti /usr/sbin/dropbearmulti.backup
echo -e "GET /openlumi/openlumi.github.io/master/files/dropbearmulti HTTP/1.0\nHost: raw.githubusercontent.com\n" | openssl s_client -quiet -connect raw.githubusercontent.com:443 -servername raw.githubusercontent.com 2>/dev/null | sed '1,/^\r$/d' > /usr/sbin/dropbearmulti
chmod +x /usr/sbin/dropbearmulti

but still does not work...
Gateway connected to the network. I can pin from Windows but I cannot do the scpfor the backup created
Thanks for supporting
BR

devbis, 2021-11-23:
What is your hardware? Post logs from console on dropbear starting?

ralocar, 2021-11-23:
Hi devbis,
Thanks for supporting
My gateway is the DGNWG05LM
When I execute the 3 lines above... no answer from the console
Please let me know if you need more info

BR

devbis, 2021-11-23:
And did you set up dropbear to start?

`/etc/init.d/dropbear start &`

Is there any response?

ralocar, 2021-11-24:
f**ck!! Thanks! It works

closed 2021-11-24 by ralocar