openlumi/openlumi.github.io #34: kernel panic while trying to return to stock fw

giltabibian, 2021-12-01:
[stoc fw update log.txt](https://github.com/openlumi/openlumi.github.io/files/7637298/stoc.fw.update.log.txt)
I've tried to flash the fw stock & restore to default with mi home app.
connected both usb2uart & usb using the mfgtools, seems like the flashing process succeeded but after rebooting the gateway i got kernel panic.
attached the log.
can anyone help me? Does someone know what can cause that issue?

giltabibian, 2021-12-01:
also attaching here the uboot enviroment settings:

Lumi iMX6ULL # pri
baudrate=115200
board_name=Xiaomi Lumi Gateway
board_rev=DGNWG05LM
bootargs=console=ttymxc0,115200 ubi.mtd=3 ubi.block=0,rootfs root=/dev/ubiblock0_0 rootfstype=squashfs cma=32M mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(ubi)
bootargs_squashfs=console=ttymxc0,115200 ubi.mtd=3 ubi.block=0,rootfs root=/dev/ubiblock0_0 rootfstype=squashfs cma=32M mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(ubi)
bootargs_ubi=console=ttymxc0,115200 ubi.mtd=3 root=ubi0:ubi rootfstype=ubifs cma=32M mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(ubi)
bootcmd=nand read ${loadaddr} 0x300000 0x700000;nand read ${fdt_addr} 0xA00000 0x100000;bootz ${loadaddr} - ${fdt_addr};
bootcmd_mfg=run mfgtool_args;bootz ${loadaddr} ${initrd_addr} ${fdt_addr};
bootdelay=1
button=released
color_after_uboot=0
color_uboot=0x111111
fdt_addr=0x83000000
fdtcontroladdr=8ef45f98
initrd_addr=0x83800000
initrd_high=0xffffffff
loadaddr=0x80800000
mfgtool_args=setenv bootargs console=ttymxc0,115200 cma=32M rdinit=/linuxrc g_mass_storage.stall=0 g_mass_storage.removable=1 g_mass_storage.idVendor=0x066F g_mass_storage.idProduct=0x37FF g_mass_storage.iSerialNumber="" mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(ubi)
mtdids=nand0=gpmi-nand
mtdparts=mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(ubi)
stderr=serial@2020000
stdin=serial@2020000
stdout=serial@2020000

Environment size: 1347/131068 bytes

devbis, 2021-12-02:
Hi, that's a known issue. The problem is that OpenWrt rewrites the initial uboot to support save envs and switch off the LEDs correctly. The revert archive is outdated, 'cause it was created for OpenWrt 19.07
You can try a couple of approaches:
- Change bootargs in uboot to the ones from stock:
```
setenv bootargs 'console=ttymxc0,115200 ubi.mtd=3 root=ubi0:rootfs rootfstype=ubifs cma=96M mtdparts=gpmi-nand:3m(boot),7m(kernel),1m(dtb),-(rootfs)
saveenv
boot
```
- Flash uboot from the stock firmware:

Add the following lines to ucl2.xml before burning the kernel

```
        <!--burn the uboot to NAND: -->
        <CMD state="Updater" type="push" body="$ mount -t debugfs debugfs /sys/kernel/debug">Mounting debugfs</CMD>
        <CMD state="Updater" type="push" body="$ flash_erase /dev/mtd0 0 0">Erasing Boot partition</CMD>
        <CMD state="Updater" type="push" body="send" file="files/u-boot-small.imx" ifdev="MX6ULL">Sending u-boot.bin</CMD>
        <CMD state="Updater" type="push" body="$ kobs-ng init -x -v --chip_0_device_path=/dev/mtd0 $FILE">Flashing Bootloader</CMD>
```

And put u-boot-small.imx to the files/ from firmware/

giltabibian, 2021-12-02:
@devbis one more question - after loading the original FW, i'm using MI home app, and it seems like all my old zigbee child devices didn't automatically connect.
tried to reconnect them and also configure them from scratch but without success.

i did this command and connected the antennas:
touch /home/root/need_update_coordinator.tag

anything else I'm missing here?

devbis, 2021-12-02:
You need to erase zigbee internal memory (PDM) to make old firmware use it. 
There are no tools in the original firmware that does it. But you can try a couple of commands that work on OpenWrt (I'm not sure all commands exist in the stock firmware):

```
stty -F "/dev/ttymxc1" raw 115200
printf '\x01\x02\x10\x12\x02\x10\x02\x10\x12\x03' > /dev/ttymxc1
```

It is just an extraction from here https://github.com/openlumi/openwrt/blob/lumi-21.02/package/utils/jntools/files/jntool

You can try hexdump as well in the background to view the chip response, if the command is executing correctly

giltabibian, 2021-12-02:
thanks for the great help mate!
it's working now :)

closed 2021-12-02 by giltabibian