openlumi/openlumi.github.io #35: Python build without Bluetooth support

ilkondr, 2021-12-13:
Hello everyone,
I tried to install passive BLE monitor for HASS, but it does not work since python build with no Bluetooth support. 
Is it supposed to be so or python may be updated for bluetooth support?
Logs are below:
`Mon Dec 13 08:12:37 2021 daemon.err hass[2782]:     family=socket.AF_BLUETOOTH, type=socket.SOCK_RAW, proto=socket.BTPROTO_HCI
Mon Dec 13 08:12:37 2021 daemon.err hass[2782]: AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'`

devbis, 2021-12-13:
Python is compiled without bluetooth support. You can build python package with bt support manually.

For Xiaomi Gateway in the custom feed python is built with bt support https://openlumi.github.io/releases/21.02.1/packages/arm_cortex-a9_neon/packages/python3-base_3.9.7-2_arm_cortex-a9_neon.ipk

devbis, 2022-01-20:
Here is the latest version of python3 from the 21.02 feed, compiled with Bluetooth support.

https://github.com/DivanX10/Openwrt-scripts-for-gateway-zhwg11lm/raw/main/packages/python3-light_3.9.9-2_arm_cortex-a9_neon_for_ble_monitor.ipk

closed 2022-01-20 by devbis