openlumi/openlumi.github.io #36: Problem installing snapclient on Openwrt version 21.02

kuenkin, 2021-12-19:
Hello,
First, thank you very much for this beautiful hack. 
I have been using it for several months and it's nice to get rid of xiaomi firm and use openwrt instead.
Today I update the firmware from 19.07 to 21.02 and I can't make work again snapclient as I had it working, flawlessly, on the previous version 19.07.
This is the error I get:
<code>tmp# snapclient
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE7compareERKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6insertEjRKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZSt7getlineIcSt11char_tr
aitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_st
ringIS4_S5_T1_EE: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE5eraseEN9__gnu_cxx17__normal_iterat
orIPKcS4_EES9_: symbol not found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE7compareEPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcS5_S5_: symbol no
t found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE6substrEjj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE10_M_replaceEjjPKcj: symbol not fou
nd
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE7reserveEj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEC1ERKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1119basic_istr
ingstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE9_M_assignERKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEC1EOS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1118basic_stri
ngstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6insertEjPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1118basic_stri
ngstreamIcSt11char_traitsIcESaIcEEC1ESt13_Ios_Openmode: symbol
not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE9push_backEc: symbol not found
Error relocating /usr/bin/snapclient: _ZStlsIcSt11char_traitsIc
ESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringI
S4_S5_T1_EE: symbol not found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE4findEPKcj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEpLEPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE4findEcj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE12_M_constructEjc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_: symbol no
t found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEaSEOS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE10_M_disposeEv: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt13runtime_errorC2ERK
NSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol no
t found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEpLERKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEEaSEPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE4findERKS4_j: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE7replaceEjjPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1115basic_stri
ngbufIcSt11char_traitsIcESaIcEE3strERKNS_12basic_stringIcS2_S3_
EE: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt12out_of_rangeC1ERKN
St7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol not
 found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1115basic_str
ingbufIcSt11char_traitsIcESaIcEE3strEv: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1118basic_stri
ngstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3
_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6resizeEjc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1119basic_istr
ingstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S
3_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE7replaceEjjRKS4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE9_M_createERjj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt16invalid_argumentC1
ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE: symbol
 not found
Error relocating /usr/bin/snapclient: _ZNKSt7__cxx1112basic_str
ingIcSt11char_traitsIcESaIcEE5rfindEcj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6resizeEj: symbol not found
Error relocating /usr/bin/snapclient: _ZSt7getlineIcSt11char_tr
aitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_st
ringIS4_S5_T1_EES4_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6insertEN9__gnu_cxx17__normal_itera
torIPKcS4_EEc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE7replaceEN9__gnu_cxx17__normal_iter
atorIPKcS4_EES9_S9_S9_: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6appendEPKc: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6appendEPKcj: symbol not found
Error relocating /usr/bin/snapclient: _ZNSt7__cxx1112basic_stri
ngIcSt11char_traitsIcESaIcEE6appendERKS4_: symbol not found
root@Altavoz_Zigbee:/tmp#</code>

The question would be can I downgrade to the previous OpenWrt version?

devbis, 2021-12-20:
Hi! This ipk, I suppose, was built for OpenWrt 19.07. 21.02 has other library versions and it fails. I guess, there is no prebuilt version for 21.02, so you have to compile it from sources.

devbis, 2021-12-20:
To downgrade to 19.07 you can use scripts from https://github.com/DivanX10/Openwrt-scripts-for-gateway-dgnwg05lm

definitio, 2021-12-22:
Snapcast 0.26.0 for OpenWrt 21.02:
[snapcast 0.26.0.zip](https://github.com/openlumi/openlumi.github.io/files/7764618/snapcast.0.26.0.zip)

closed 2022-01-13 by devbis