openlumi/openlumi.github.io #39: Problem after root

Pmoshbr, 2022-01-20:
After I did the root process and reboot the device I received this message at serial terminal (normal boot):

```
INIT: version 2.88 booting
Starting udev
[    3.866975] udevd[105]: starting version 3.1.5
[    4.387675] UBIFS (ubi0:0): completing deferred recovery
[    4.501123] UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 129
[    4.533813] UBIFS (ubi0:0): deferred recovery completed
INIT: Entering runlevel: 5
**Configuring network interfaces... eth0: ERROR while getting interface flags: No such device**
Running local boot scripts (/etc/rc.local).
umount: /mnt/.psplash: not mounted

Freescale i.MX Release Distro 4.1.15-2.0.0 imx6ull14x14evk /dev/ttymxc0

imx6ull14x14evk login:
```

What I did wrong?? I did not made the backup yet... can I forget the backup process and flash openwrt with serial??

devbis, 2022-01-20:
It is a normal login prompt for Unixes. Use root as a user and your password to login to bash.

Pmoshbr, 2022-01-20:
Yes I did the login but still without a network...
After login I managed to copy /etc/rc.local.mi to /etc/rc.local and add dropbear line.. And run as root /etc/rc.local...
Then everything comes back to normal, the ko file driver for rtl chip was loaded and the network was up again... I finished the update with success...

Thanks

closed 2022-01-20 by Pmoshbr