openlumi/openlumi.github.io #41: Stuck after rebooting

ljovcheg, 2022-01-30:
I don't have much knowledge with bash, so I will write all my steps.
Tried to get root for my DGNWG05LM with uart.
Paused loading, inserted ` setenv bootargs "${bootargs} single rw init=/bin/bash" && boot`, used `passwd` to set new password.
Restarted gateway and after loading got INIT: no more processes left in this runlevel.
Found a command that could help me: `sed -i "s/#mxc0/mxc0/" /etc/inittab`. Again, as my knowledge is bad, I started all over with setenv bootargs "${bootargs}... and passwd.
Entered inittab command and rebooted device. Loading finished with **Enter runlevel**
When entering any number I get 

```
INIT: Entering runlevel: _entered_number_
INIT: no more processes left in this runlevel
```

Found how to check file sizes with bash and found out, that inittab size is 0. I don't know if this is important info.  
Is it brick now? 
Thanks

devbis, 2022-01-31:
```
# /etc/inittab: init(8) configuration.
# $Id: inittab,v 1.91 2002/01/25 13:35:21 miquels Exp $

# The default runlevel.
id:5:initdefault:

# Boot-time system configuration/initialization script.
# This is run first except when booting in emergency (-b) mode.
si::sysinit:/etc/init.d/rcS

# What to do in single-user mode.
~~:S:wait:/sbin/sulogin

# /etc/init.d executes the S and K scripts upon change
# of runlevel.
#
# Runlevel 0 is halt.
# Runlevel 1 is single-user.
# Runlevels 2-5 are multi-user.
# Runlevel 6 is reboot.

l0:0:wait:/etc/init.d/rc 0
l1:1:wait:/etc/init.d/rc 1
l2:2:wait:/etc/init.d/rc 2
l3:3:wait:/etc/init.d/rc 3
l4:4:wait:/etc/init.d/rc 4
l5:5:wait:/etc/init.d/rc 5
l6:6:wait:/etc/init.d/rc 6
# Normally not reached, but fallthrough in case of emergency.
z6:6:respawn:/sbin/sulogin
mxc0:12345:respawn:/bin/start_getty 115200 ttymxc0
# /sbin/getty invocations for the runlevels.
#
# The "id" field MUST be the same as the last
# characters of the device (after "tty").
#
# Format:
#  <id>:<runlevels>:<action>:<process>
#

1:12345:respawn:/sbin/getty 38400 tty1
```

You can populate the file with the following content to fix the issue from single-mode bash.

ljovcheg, 2022-01-31:
Thanks a lot, it helped.

closed 2022-01-31 by ljovcheg