openlumi/openlumi.github.io #44: The ssh server won't start.

FloyFire, 2022-10-03:
Hello! Following the instructions, I got root rights, but the ssh server could not start. After adding the line "/etc/init.d/dropbear start &" to the file " /etc/rc.local" and rebooting, nothing changed. Running the command "/etc/init.d/dropbear start" does not work either. 

Also, tried with and without this command: "sed -i "s/#mxc0/mxc0/" /etc/inittab"

Earlier i accidentally managed to start ssh server 1 time somehow, using chaotic commands :D

Model: Aqura Hub: ZHWG11LM V2 (looks exactly like in the manual).

Please, if you can, help.
P.S. Sorry for the English, using translator.

devbis, 2022-10-03:
https://openlumi.github.io/gain_root.html#gaining-root
Did you try updating `dropbearmulti` from the lines in the first paragraph? 
Probably, you have dropbear with missing ssh server support.

FloyFire, 2022-10-03:
Thank you for replying! Unfortunately, yes. I have tried every possible option. The command "/etc/init.d/dropbear start" has no effect on the gateway. 

Used and this command: "sed -i "s/#mxc0/mxc0/" /etc/inittab"
and this
"mv /usr/sbin/......", "echo -e "GET /openlumi/......", "chmod +x /usr/....."

This inscription never happened: "Starting Dropbear SSH server: Pseudo-terminal will not be allocated because stdin is not a terminal."

If you need to send something from the data, please tell me about it.

devbis, 2022-10-06:
Can you see the process running? 
`ps aux | grep dropbear`

FloyFire, 2022-10-06:
Dear devbis, thanks for the feedback! I ran the command and got this result:
[image]

`root@imx6ull14x14evk:~# ps aux | grep dropbear`

`root       838  0.0  0.1   1872   332 ttymxc0  S+   23:16   0:00 grep dropbear`

If I understand correctly, this command tells the terminal if dropbear is running?

P.S. I'm not sure if this information is useful, but I have used the ./reset.sh command a couple of times to reset the device to its original settings.

devbis, 2022-10-07:
Nope, it doesn't run at the time of screenshot, means the problem is with the start.
You could try to run it manually, like `dropbear -R -F -E` but I'm not sure about the correct flags. Probably clues are in `/etc/init.d/dropbear`

FloyFire, 2022-10-07:
Dear devbis! When I entered the command: dropbear -R -F -E, it seemed to start.

 
[image]

`root@imx6ull14x14evk:/home# dropbear -R -F -E`
`[9425] Oct 07 15:21:32 Not backgrounding`

If I understand correctly, "not backgrounding" just says that it is not a background process, but it's ok.

Tried to connect, it came up with this: 

[image]
`[9435] Oct 07 15:21:37 Child connection from 192.168.1.134:33120`
`[9435] Oct 07 15:21:37 Couldn't create new file /etc/dropbear/dropbear_ecdsa_host_key.tmp9435: No such file or directory`
`[9435] Oct 07 15:21:37 Exit before auth: Couldn't read or generate hostkey /etc/dropbear/dropbear_ecdsa_host_key`

Also, I tried to create a key with the command: dropbearkey -f ~/.ssh/id_dropbear -t rsa -s 2048. It didn't work.

devbis, 2022-10-07:
Does `/etc/dropbear/` exist? If not, create it with `mkdir -p /etc/dropbear`

FloyFire, 2022-10-07:
Thanks a lot, devbis! SSH is working :)

closed 2022-10-07 by FloyFire