openlumi/openlumi.github.io #45: Error on rooting

rvgeex, 2022-10-09:
Hi,
I have a problem when I try to root the DGNWG05LM.
After use the command _setenv bootargs "${bootargs} single rw init=/bin/bash" && boot_,  I have a error message:
[image]

```
[    2.870402] VFS: Mounted root (ubifs filesystem) on device 0:14.
[    2.877693] devtmpfs: mounted
[    2.881892] Freeing unused kernel memory: 540K (80aa1000 - 80b28000)
bash: cannot set terminal process group (-1): Inappropriate ioctl for device
bash: no job control in this shell
bash-4.3#
```
If I don't use init=/bin/bash, the boot goes further but it doesn't work anymore
```
=> setenv bootargs "${bootargs} single rw" && boot

NAND read: device 0 offset 0x300000, size 0x700000
 7340032 bytes read: OK

NAND read: device 0 offset 0xa00000, size 0x100000
 1048576 bytes read: OK
Kernel image @ 0x80800000 [ 0x000000 - 0x617290 ]
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
```
...
```
[    2.876859] devtmpfs: mounted
[    2.880976] Freeing unused kernel memory: 540K (80aa1000 - 80b28000)
INIT: version 2.88 booting
Starting udev
cmp: EOF on /etc/udev/cache.data
udev: Not using udev cache because of changes detected in the following files:
udev:     /proc/version /proc/cmdline /proc/devices
udev:     lib/udev/rules.d/* etc/udev/rules.d/*
udev: The udev cache will be regenerated. To identify the detected changes,
udev: compare the cached sysconf at   /etc/udev/cache.data
udev: against the current sysconf at  /dev/shm/udev.cache
[    3.869683] udevd[105]: starting version 3.1.5
Populating dev cache
Give root password for maintenance
(or press Control-D to continue):

```
Could you help me?

devbis, 2022-10-10:
And what prevents you from setting a root password in bash?

rvgeex, 2022-10-10:
If I don't use init=/bin/bash, it asks me to enter a password (which I don't know) not to set a new one.

rvgeex, 2022-10-10:
> You must run with init=/bin/bash, your gif shows you booted normally.

But after the boot, I can't write anything

devbis, 2022-10-10:
It could be a problem with your usb-serial convertor, computer, or OS.

closed 2024-01-05 by devbis