openlumi/openlumi.github.io #48: Gateway seems to crash after booting

bronks75, 2022-11-19:
Hello,
I have an issue when I boot with bash, the system seems to crash after few seconds.
I can launch 1 command if i go fast but after that no more life.
Can you please help me on this ?

U-Boot 2016.03 (Nov 07 2017 - 20:11:55 +0800)

CPU:   Freescale i.MX6ULL rev1.1 528 MHz (running at 396 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 34C
Reset cause: POR
Board: Lumi international Gateway
I2C:   ready
DRAM:  256 MiB
NAND:  256 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

Display: TFT43AB (480x272)
Video: LCDIF@0x21c8000 is fused, disable it
In:    serial
Out:   serial
Err:   serial
Net:   Ethernet@0x20b4000 is fused, disable it
Board Net Initialization Failed
No ethernet found.

## Enter main_loop() Now##
Normal Boot
Hit any key to stop autoboot:  0
=> setenv bootargs "${bootargs} single rw init=/bin/bash" && boot

NAND read: device 0 offset 0x300000, size 0x700000
 7340032 bytes read: OK

NAND read: device 0 offset 0xa00000, size 0x100000
 1048576 bytes read: OK
Kernel image @ 0x80800000 [ 0x000000 - 0x617290 ]
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300c0e1
Modify /soc/aips-bus@02000000/tsc@02040000:status disabled
Modify /soc/aips-bus@02000000/can@02090000:status disabled
Modify /soc/aips-bus@02000000/can@02094000:status disabled
Modify /soc/aips-bus@02100000/lcdif@021c8000:status disabled
Modify /soc/aips-bus@02100000/pxp@021cc000:status disabled
Modify /soc/aips-bus@02100000/csi@021c4000:status disabled
Modify /soc/aips-bus@02000000/ethernet@020b4000:status disabled
Modify /soc/aips-bus@02100000/usb@02184200:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/sai@0202c000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/sai@02030000:status disabled
Modify /soc/aips-bus@02100000/serial@021f4000:status disabled
Modify /soc/aips-bus@02100000/serial@021fc000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/serial@02018000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f0000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f4000:status disabled
Modify /soc/aips-bus@02000000/pwm@020f8000:status disabled
Modify /soc/aips-bus@02000000/pwm@020fc000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/ecspi@02010000:status disabled
Modify /soc/aips-bus@02000000/spba-bus@02000000/ecspi@02014000:status disabled
Modify /soc/aips-bus@02100000/i2c@021a8000:status disabled
Modify /soc/aips-bus@02100000/i2c@021f8000:status disabled
Modify /soc/aips-bus@02000000/gpt@020e8000:status disabled
Modify /soc/aips-bus@02000000/epit@020d4000:status disabled
ft_system_setup for mx6
No PMIC found!

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15+gb63f3f5 (chenlong@gitlab) (gcc version 5.3.0 (GCC) ) #37 SMP PREEMPT Tue Jun 19 15:28:38 CST 2018
[... 212 lines cut ...]
[    2.645712] dhd_module_init in
[    2.651110] input: regulators:gpio-keys as /devices/platform/regulators/regulators:gpio-keys/input/input1
[    2.662216] snvs_rtc 20cc000.snvs:snvs-rtc-lp: setting system clock to 1970-01-01 01:01:14 UTC (3674)
[    2.693439] gpio_dvfs: disabling
[    2.696728] vref-3v3: disabling
[    2.699891] can-3v3: disabling
[    2.703447] ALSA device list:
[    2.706449]   #0: wm8524-audio
[    2.712011] UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 74
[    2.740458] UBIFS (ubi0:0): recovery needed
[    2.822768] UBIFS (ubi0:0): recovery completed
[    2.827489] UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "rootfs"
[    2.835112] UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[    2.845184] UBIFS (ubi0:0): FS size: 241635328 bytes (230 MiB, 1903 LEBs), journal size 12062720 bytes (11 MiB, 95 LEBs)
[    2.856126] UBIFS (ubi0:0): reserved for root: 4952683 bytes (4836 KiB)
[    2.862809] UBIFS (ubi0:0): media format: w4/r0 (latest is w4/r0), UUID AD7DD6E8-1329-4760-93D9-E1A4D3A2465D, small LPT model
[    2.875025] VFS: Mounted root (ubifs filesystem) on device 0:14.
[    2.882132] devtmpfs: mounted
[    2.886251] Freeing unused kernel memory: 540K (80aa1000 - 80b28000)
bash: cannot set terminal process group (-1): Inappropriate ioctl for device
bash: no job control in this shell
bash-4.3# ls
bin   dev  home  lumi   mnt  proc  sbin  tmp  var
boot  etc  lib   media  opt  run   sys   usr  wpa
bash-4.3#

devbis, 2022-11-20:
Do you have a log for crash? The log above seems to be booting fine.

bronks75, 2022-11-20:
Thanks for your answer.
How can I do that, some file to tail (I can launch 1 command before crash) ?

devbis, 2022-11-29:
You could try to change usb-ttl adapter or the computer. It could be the problem not with the gateway itself.

closed 2024-01-05 by devbis