openlumi/openlumi.github.io #49: After intall OpenWRT 21 Xiaomi Hub not booting

Viper7000, 2023-01-08:
Tried to install the latest version. Today the script install 21.02.5
Nothing succeeded. Everything seems to go right, after installation, the hub is rebooted and the boot ends with the line "reboot: System halted"
I took another script that install 19.07.10 - the same result. The boot log is different, but it doesn't load. One of the installations ended with a cyclic reboot.
I also tried using uuu and mfgtools - it doesn't boot into any!
As a result, I returned the stock through mfgtools - everything works fine. Both Serial and USB are already soldered...
The gateway has Mi firmware 3.4.6_132. Suggests 3.5.9_004. Haven't updated yet...

Serial log: Stoc Mi boot - Install OWRT21 - boot after install
[stoc_OWRT_halt.txt](https://github.com/openlumi/openlumi.github.io/files/10369148/stoc_OWRT_halt.txt)

Viper7000, 2023-01-09:
This is the firmware log from working Mi stoc to OpenWTR 19.07.7 via mfgtools. After OpenWRT start, cyclic loading

Install OWRT21 - boot after install
[stoc_mfgtools-19.txt](https://github.com/openlumi/openlumi.github.io/files/10372146/stoc_mfgtools-19.txt)

Viper7000, 2023-01-09:
Try to install firmware by UART via ZOC terminal. Write all partitions and reboot. And I have boot loop.
When write in log Reset cause: WDOG

Start after UART firmware log
[OWRT21-UART-ZOC-start-loop.txt](https://github.com/openlumi/openlumi.github.io/files/10375201/OWRT21-UART-ZOC-start-loop.txt)

Viper7000, 2023-01-12:
All the problems turned out to be due to the stock u-boot 2016. Because of it, new versions were not installed. Thanks **Alx** for the help in this question. The installation script does not flash a new u-boot. Possible script error.
I install a new u-boot manually according to the instructions 
https://github.com/openlumi/openwrt/releases/tag/v21.02.0-20210902
After that, everything is installed.
thanks for the help @Alx2000y, @devbis, @G1K

closed 2023-01-12 by Viper7000

Viper7000, 2023-01-12:
Here I was wrong! The new script 21.02.5 flashes u-boot, but firmware still does not work. There is already a problem not in u-boot. In any case, in manual mode, according to the instructions above, everything is set. True, in my particular case, the firmware did not have time to load and reboot by WDT. Therefore, I did for myself a dbt file with a disabled WDT. I'll leave the converted files here for those who understand what to do with them
[openlumi-21.02.5-imx6-imx6ull-xiaomi-dgnwg05lm_nowd.zip](https://github.com/openlumi/openlumi.github.io/files/10402245/openlumi-21.02.5-imx6-imx6ull-xiaomi-dgnwg05lm_nowd.zip)
[openlumi-22.03.2-imx-cortexa7-imx6ull-xiaomi-dgnwg05lm_nowd.zip](https://github.com/openlumi/openlumi.github.io/files/10402252/openlumi-22.03.2-imx-cortexa7-imx6ull-xiaomi-dgnwg05lm_nowd.zip)

Viper7000, 2023-01-17:
Now I'm testing firmware OpenWRT 22.03.2 on Xiaomi Gateway DGNWG05LM from here https://openlumi.github.io/releases/22.03.2/targets/imx/cortexa7/

Only dtb took my 22.03.2without WDT, the one that is fixed above.
I will post my test results here. But remember that I have WDT disabled and my particular case may differ from yours.

Installing packages **mc, nano, htop** from luci - no problem

1. **lumimqtt** - install and works without problems. Generally satisfied
2. **zigbee2MQTT** seems to work. So far I have installed the latest Zigate firmware on 115200 (https://github.com/openlumi/ZiGate/releases/tag/60a8--20220808-1857). I took a couple of test Aqara vibration sensors (DJT11LM) - they seem to connected and work. Cooler than in Mi! While there is nothing more to check with
3. **MPD** does not want to work. I start with HA radio, I see incoming traffic on luci, but there is no sound! It feels like there is no sound. Periodically I do not see MPD in HA! Didn't understand. On version 19 it worked immediately and the radio started
4. **ble2mqtt** doesn't work. At the same time, some mqtt sends from the devices, but there is no data from the sensors. Issue created on Github.(https://github.com/devbis/ble2mqtt/issues/47) While turned off
5. **ESPruinoHab** - works, sends data from the sensor. Basically functional! But devices in HA Discovery are registered very primitively! There is no firmware, no model, no device manufacturer. There is nothing on the device, but there is data from the sensors!
6. Put Russian on luci - somehow more familiar

Viper7000, 2023-01-18:
I managed to start **MPD** on 22.03.2
Remove the `mpd-full` package if necessary. Now download from version 21.02.5:
https://downloads.openwrt.org/releases/21.02.5/packages/arm_cortex-a7_neon-vfpv4/packages/libnpupnp_4.1.1-1_arm_cortex-a7_neon-vfpv4.ipk
https://downloads.openwrt.org/releases/21.02.5/packages/arm_cortex-a7_neon-vfpv4/packages/mpd-full_0.21.26-1_arm_cortex-a7_neon-vfpv4.ipk
Now we install `libnpupnp` first, then `mpd-full`. I installed in luci via "Upload package"
Check the settings, restart the service. Working

**PS** 
I use a `Home assistant`.
After rebooting the gateway, the Home assistant does not immediately find the **MPD**. Appears after about 3-5 minutes. Although the port on the gateway is open for connection. Didn't understand why

Viper7000, 2023-01-18:
Checked the **network settings reset** on 22.03.2
I hold the button for 10-12 seconds, release it - blinks yellow 3 times
All networks are down! 3 min passed - nothing (reboot doesn't work!)
I reboot with the power off. Loading 2-3 minutes (well, in my case) - `OpenWRT` network appears
Then everything is according to the instructions. The root password remains the same as you set

Checked the **factory reset** on 22.03.2
I hold the button for 20-22 seconds, release it - Approximately 5-6 seconds (erases the software), blinks red 3 times
All networks are down! 3 min passed - nothing (reboot doesn't work!)
I reboot with the power off. Loading 70-80 sec (well, in my case) - `OpenWRT` network appears
Then everything is according to the instructions. The root password is reset to blank!

flavio-simonelli, 2023-03-14:
Hi @Viper7000 , I was also trying to install openwrt on my hub identical to yours and I get the same errors as you. Could you explain me better with a few more details the procedure to start open wrt on my hub? A thousand thanks

devbis, 2023-03-15:
You should solder USB and try flashing DTB without watchdog (Viper7000 attached it to the comment above)
Use UUU or mfgtools for it.

kuenkin, 2023-08-25:
Hello,
I'm on the same boat but, with the ZHWG11LM version.
Stock works (not interessted) and I can flash it through mfgtools.
If I try to go to OpenWrt through the OTA method it leads to system halted.
Flashing with mfgtools Openwrt 19 leads to cycle rebooting due to the watchdog.
Flashing with uuu Openwrt 21 gives the same result, the watchdog reboots the system before finishing.
One thing I noticed is that the board is identified, incorrectly, as DGNWG05LM, don't know if is related to the problem.
In the log I also can spot an error mounting rootfs, I will update later with a log.
I guess that the next step should be to use dtc to make a dtb without watchdog for the ZHWG11LM, but maybe there should be other way...

devbis, 2023-08-28:
You may try to flash the system with replacing DTB with the same but with disabled watchdog. It helped in several similar cases. This one is for Aqara:
[imx6ull-aqara-zhwg11lm.dtb.zip](https://github.com/openlumi/openlumi.github.io/files/12453143/imx6ull-aqara-zhwg11lm.dtb.zip)

kuenkin, 2023-08-29:
Thank you devbis, it works and can do sysupgrade with this dtb without watchdog as Viper7000 stated already.
The only downside, also mentioned, is that it doesn't reboots, but much better than the stock, so thanks again!!!