openlumi/openlumi.github.io #50: wifi problem

Spp595, 2023-04-04:
Когда я ввожу команду `bash-4.3# /lumi/app/miio/wifi_start.sh `
пишется следующее: 
```
bash-4.3# /lumi/app/miio/wifi_start.sh 
ip
Keenetic-595 my-password
wlan0: ERROR while getting interface flags: No such device
killall: /proc lacks process entries (not mounted ?)
wlan0: ERROR while getting interface flags: No such device
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; No such device.
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
udhcpc: SIOCGIFINDEX: No such device
get ip addr :
Warning: cannot open /proc/net/dev (No such file or directory). Limited output.
```
Что не так?

devbis, 2023-04-04:
Probably, you're trying to run the script from single mode. with init=/bin/bash

Just set root password there and unblock getty. Other commands you have to run in normal mode.

Spp595, 2023-04-04:
> Вероятно, вы пытаетесь запустить скрипт из одиночного режима. с помощью init=/bin/bash
> 
> Просто установите там пароль root и разблокируйте getty. Другие команды вы должны выполнять в обычном режиме.

Это нужно выполнять после каждого перезапуска?

closed 2023-04-05 by Spp595