openlumi/openlumi.github.io #56: MacOs Ventura

leovj, 2024-05-26:
I performed the procedure from Macos Ventura 13.6.7. the gateway is in resetting mode ... but when I enter the command 

uuu aqara_zhwg11lm.uuu 

from the terminal it returns:

Wait for Known USB Device Appear.. 

Can you give me some solution?

devbis, 2024-05-26:
Probably you don't have permissions. Try run uuu with sudo

leovj, 2024-05-27:
iMac-di-Leonardo:uuu-openwrt-23.05.2 leo$ sudo uuu aqara_zhwg11lm.uuu
Password:
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.165-0-g7347a80

Wait for Known USB Device Appear...

leovj, 2024-05-27:
U-Boot 2020.04 (Oct 24 2021 - 09:01:35 +0000)

CPU:   Freescale i.MX6ULL rev1.1 528 MHz (running at 396 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 35C
Reset cause: POR
Model: Xiaomi Lumi Gateway
Board: DGNWG05LM
DRAM:  256 MiB
NAND:  256 MiB
MMC:   FSL_SDHC: 0
Loading Environment from NAND... *** Warning - bad CRC, using default environment

In:    serial@2020000
Out:   serial@2020000
Err:   serial@2020000
Net:   CPU Net Initialization Failed
No ethernet found.
Hit any key to stop autoboot:  0 
Lumi iMX6ULL #

devbis, 2024-05-28:
Did you solder all 7 wires? 3 for uart and 4 for USB?

leovj, 2024-06-01:
> Did you solder all 7 wires? 3 for uart and 4 for USB?

Yes!

leovj, 2024-06-01:
I did something really stupid, I tried to update the system from openwrt, probably loading the wrong ISO. The hub now restarts continuously and I would like to return everything to its previous state. But I can not. With the procedure it seems that it doesn't "feel" the USB port..