openlumi/openlumi.github.io #57: Zesp32 not load

felixortegam, 2024-12-07:
I have installed zesp32 following the tutorial, but it doesn't load. I ran the execution from the app directory to check for errors, and here is the result:

[image]

I have a DGNWG05LM with OpenLumi installed, version OpenLumi 23.05.4 r24012-d8dd03c46f / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38.

Thanks!

devbis, 2024-12-07:
It might happen that zesp32 is abandoned. The instruction was written several years ago

felixortegam, 2024-12-08:
Ups! Any good alternative? thanks!

devbis, 2024-12-09:
It depends on your needs. You can use [home assistant](https://github.com/openlumi/homeassistant_on_openwrt/), zigbee2mqtt, domoticz, [homed](https://wiki.homed.dev/page/ZigBee).

felixortegam, 2024-12-10:
OK, i have installed lumimqtt and set the zigbee firmware on router mode. Works great with HA in my homelab. Thanks!

closed 2024-12-11 by devbis